jneilliii/OctoPrint-PrintScheduler #1: ghost scrolling when editing the schedule date

markdavidjohnson, 2021-08-23:
When editing the start at time, if using the popout calendar and its time scrollers, the times have a residual scroll event after scrolling is completed. After scrolling a lot I saw the minute scroller move continuously by itself. Repeatable.

I also use the GPIO Control plugin to turn the actual printer on. I'd love to include in the schedule a GPIO set to on command. Is there a good way to do this?

jneilliii, 2021-08-23:
I'm only using standard html input tags of type `datetime-local`, so this to me almost sounds like a browser related implementation issue, or it just doesn't quite work as expected in general. Which browser are you using, as I'm finding out now that all browsers do not support that type. 

https://www.w3schools.com/tags/att_input_type_datetime-local.asp

markdavidjohnson, 2021-08-23:
Oh nice, I did't know about that input tag. So testing using the w3schools link, i get the same single fantom scroll (looks like it is just trying to not display any partial times for some reason) but i wasnt able to get it to do the multiple fantom scrolls. It's harder to get that fail mode to manifest though, so i wouldn't say it only happens in the plugin. I do think youre right though. At least the single fantom scroll is a bigger issue, and the multiple fantom scrolling is also probably not your code. Thanks for checking!

If you have any thoughts on the GPIO integration I'd love to hear. My hope today was to have it start 3 minutes before my alarm clock this morning, and I could just check that it was going before getting ready (or snoozing), but I didn't want to leave the fans running all night, so that use case currently doesn't work.

jneilliii, 2021-08-23:
I'm going to be looking into a different date/time selector in the near future as it turns out firefox doesn't support it at all (#3).

jneilliii, 2021-08-25:
I think I might have this working now with the above commit. You can test it by switching the plugin's release channel to `Release Candidate` in OctoPrint's software update settings and upgrade when prompted. 

as for the GPIO related question, I'm not familiar enough with that stuff, but that is the reason for the before print start options in the settings. you might be able to use that to control the GPIO via system commands.