jneilliii/OctoPrint-PrintScheduler #2: Not working for me.

Ted-Printable, 2021-08-23:
Hi, need to say this is a plugin I have been waiting for, great idea.!!

Not sure if it is operator error but I could not get it to work.
I uploaded a file.
Scheduled it to print five minutes ahead, then clicked save to save the schedule, but nothing happened when the time came to start.
I am using the latest OctoPi image and my Pi is fully updated. Any help please?

jneilliii, 2021-08-23:
One thing to double-check is that your Pi has the correct timezone and date/time set within its OS. To do this you will need to SSH to the Pi and run `sudo raspi-config` and find the localization options.

Ted-Printable, 2021-08-23:
Wow, rookie mistake, but I did assume OctoPrint will sync with my browzer.
Thank you, I will check again.

Of concern is the fact this print did start later, but it did not execute the complete file code. My pre-print g-codes that extrude a blob and also the print was placed at the origin, in stead of the middle of buildplate.
Am I doing something else wrong?

jneilliii, 2021-08-23:
The plugin just starts the file with the same commands you would use if you click the select and print button in the file list.

Ted-Printable, 2021-08-24:
Ok, many things could cause the wrong place print, especially since there was a print still on the bed.

I have changed the Pi's time zone and confirmed it to be local time.
Tried to schedule another print 5min ahead, still no response.
Big probability I am doing something wrong still, but I cannot think of anything since there is very little user interaction on this plugin required, to my knowledge.
Please help a bit more.

jneilliii, 2021-08-24:
Go into OctoPrint's logging section in settings and at the bottom select `octoprint.plugins.printscheduler` from the drop-down and set it to debug. Save and restart OctoPrint and try again. Once it doesn't hit the mark download the file octoprint.log and drag it here into a comment for analysis. Also, please let me know what browser you are using.

Ted-Printable, 2021-08-25:
Hi, Thanks very much for the help. Please allow me a bit of time to get this done. I have to bring another printer online again after numerous upgrades first.Will send the log and my observations as soon as possible. ThanksTed.  From: jneilliiiSent: Tuesday, 24 August 2021 15:45To: jneilliii/OctoPrint-PrintSchedulerCc: Ted-Printable; AuthorSubject: Re: [jneilliii/OctoPrint-PrintScheduler] Not working for me. (#2) Go into OctoPrint's logging section in settings and at the bottom select octoprint.plugins.printscheduler from the drop-down and set it to debug. Save and restart OctoPrint and try again. Once it doesn't hit the mark download the file octoprint.log and drag it here into a comment for analysis. Also, please let me know what browser you are using.

Ted-Printable, 2021-08-25:
Hi. Some feedback..... I am commissioning a upgraded Ender_3 which I play with a lot. The Raspberry_Pi4B-4Gb was loaded with the latest 64Bit version of OctoPi image, which is a test I am doing to see how stable this 64Bit is. I haven’t had time to use the plugin on a production printer again. I am using Google Chrome Browser. I loaded a test print and scheduled it 5min ahead, saved the schedule.Then realized I did not set the time-zone on this Pi so I deleted the schedule.I did not delete the file as I planned using it later.Then I set the time-zone after learning some new tricks since this version runs on Ubunto. After a reboot, the printer started printing the DELETED SHEDULED file, correctly.My concern is that the schedule was deleted but still ran. This can cause major problems in my view. So my concern is, if a mistake is made and a scheduled event is erased, it should be gone from existence. There should be no possibility of the print starting in the middle of the night unintentionally after the schedule was erased. I attached the log file as requested, hope this helps.Ted.  From: jneilliiiSent: Tuesday, 24 August 2021 15:45To: jneilliii/OctoPrint-PrintSchedulerCc: Ted-Printable; AuthorSubject: Re: [jneilliii/OctoPrint-PrintScheduler] Not working for me. (#2) Go into OctoPrint's logging section in settings and at the bottom select octoprint.plugins.printscheduler from the drop-down and set it to debug. Save and restart OctoPrint and try again. Once it doesn't hit the mark download the file octoprint.log and drag it here into a comment for analysis. Also, please let me know what browser you are using.

jneilliii, 2021-08-25:
attached files don't make it to me when sent via email, you need to drag it into a comment here on GitHub.

jneilliii, 2021-08-25:
Also, the print starting after the schedule had been cleared is definitely unexpected. Not sure what may have happened there. After you cleared the list did you also press the save button at the top?

jneilliii, 2021-08-25:
Yeah, I'll probably make it automatically save when the list is cleared. I could see how the save button popping up when cleared could cause some confusion.

Ted-Printable, 2021-08-30:
Sorry it took some time, but here is the log file
[octoprint.log](https://github.com/jneilliii/OctoPrint-PrintScheduler/files/7076118/octoprint.log)

Ted-Printable, 2021-08-30:
Hi, I used the plugin after the update.
It is working as expected, very nice thank you.
Ps: the time-setting upgrade is a improvement, like it!