jneilliii/OctoPrint-PrintScheduler #3: No calendar widget to set date and time when using Firefox

ArtSpaziano, 2021-08-23:
Using Firefox 91.0.1 (64-bit), no calendar widget to set date and time.
See attached screen shot.
If I try to enter the date and time manually, the entire entry disappears after I select 'Save'.

Octoprint Version 1.6.1  
OctoPi Version 0.18.0, running on Raspberry Pi 3 Model B Rev 1.2

'seems ok using Chrome, though.

[image]

jneilliii, 2021-08-23:
Yeah, I've recently learned that firefox doesn't support this html type. Will be looking for alternatives.

InnoProd, 2021-08-24:
> Using Firefox 91.0.1 (64-bit), no calendar widget to set date and time.
> See attached screen shot.
> If I try to enter the date and time manually, the entire entry disappears after I select 'Save'. [...]

Just discovered this bug as well. I try to avoid Chrome, so I'll be glad to see the next version come out!

jneilliii, 2021-08-24:
Yeah, I made some progress on that last night, but it's still not quite right. Will hopefully be out later tonight.

jneilliii, 2021-08-25:
I think I might have this working now with the above commit. You can test it by switching the plugin's release channel to `Release Candidate` in OctoPrint's software update settings and upgrade when prompted.

ArtSpaziano, 2021-08-25:
It works for me - at least, the calendar appears when I click inside the date/time window and I can save the entry. 
'haven't checked to see that it actually executes the schedule but this is a big improvement because FireFox is my default browser.  

Thanks for the quick turn-around on this fix!

InnoProd, 2021-09-08:
Confirmed functional, thank you!