jneilliii/OctoPrint-PrintScheduler #9: Scheduled jobs gone after refresh

sildehoop, 2021-08-28:
If I make a new scheduled print job and save then refresh the page all scheduled jobs are gone and not working. Bud when I stay on the page and wait it works.

```
2021-08-28 20:22:59,737 - octoprint.plugins.printscheduler - DEBUG - Starting repeated timer.
2021-08-28 20:22:59,982 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 6, 'printer_state': 'OPERATIONAL'}
2021-08-28 20:23:00,115 - octoprint.plugins.pluginmanager - INFO - Loaded notice data from disk, was still valid
2021-08-28 20:23:00,248 - octoprint.plugins.tracking - INFO - Sent tracking event pong, payload: {'version': '1.6.1', 'os': 'linux', 'bits': 32, 'python': '3.7.3', 'pip': '20.3.3', 'cores': 4, 'freq': 1500.0, 'ram': 1903988736, 'pi_model': 'Raspberry Pi 4 Model B Rev 1.2', 'octopi_version': '0.18.0', 'plugins': 'printscheduler:0.0.3rc1,simpleemergencystop:1.0.5,printtimegenius:2.2.8,navbartemp:0.14,floatingnavbar:0.3.7,display_eta:2.0.3,dashboard:1.18.3,bltouch:0.3.4,displaylayerprogress:1.26.0,change_filament:0.3.2,bedlevelvisualizer:1.1.0'}
2021-08-28 20:23:00,410 - octoprint.plugins.tracking - INFO - Sent tracking event startup, payload: {'version': '1.6.1', 'os': 'linux', 'bits': 32, 'python': '3.7.3', 'pip': '20.3.3', 'cores': 4, 'freq': 1500.0, 'ram': 1903988736, 'pi_model': 'Raspberry Pi 4 Model B Rev 1.2', 'octopi_version': '0.18.0'}
2021-08-28 20:23:00,526 - octoprint.server.util.flask - INFO - Passively logging in user Sil from 2001:1c04:3f0b:6500:ac79:b06:2e9f:fc88
2021-08-28 20:23:00,526 - octoprint.access.users - INFO - Logged in user: Sil
2021-08-28 20:23:07,297 - octoprint.server.util.sockjs - INFO - New connection from client: 2001:1c04:3f0b:6500:ac79:b06:2e9f:fc88
2021-08-28 20:23:07,341 - octoprint.server.util.flask - INFO - Passively logging in user Sil from 2001:1c04:3f0b:6500:ac79:b06:2e9f:fc88
2021-08-28 20:23:07,341 - octoprint.access.users - INFO - Logged in user: Sil
2021-08-28 20:23:09,769 - octoprint.server.util.sockjs - INFO - User Sil logged in on the socket from client 2001:1c04:3f0b:6500:ac79:b06:2e9f:fc88
2021-08-28 20:23:59,744 - octoprint.plugins.printscheduler - DEBUG - Looping through scheduled jobs.
2021-08-28 20:24:17,856 - octoprint.plugins.dashboard - ERROR - Frontend javascript error detected (this error is not necesarily to do with dashboard):
Message: TypeError: (element).datetimepicker is not a function. (In '(element).datetimepicker("getValue")', '(element).datetimepicker' is undefined) - URL: http://octopi.local/static/webassets/packed_plugins.js?bc317fcc - Line: 8942 - Column: 63 - Error object: {}
2021-08-28 20:24:24,075 - octoprint.plugins.printscheduler - DEBUG - Sorthing scheduled jobs from: [{'name': 'v3_boat_box.gcode', 'path': 'v3_boat_box.gcode', 'start_at': '2021-08-28 20:25'}].
2021-08-28 20:24:24,075 - octoprint.plugins.printscheduler - DEBUG - Sorting scheduled jobs to: [{'name': 'v3_boat_box.gcode', 'path': 'v3_boat_box.gcode', 'start_at': '2021-08-28 20:25'}].
2021-08-28 20:24:24,969 - octoprint.plugins.dashboard - ERROR - Frontend javascript error detected (this error is not necesarily to do with dashboard):
Message: TypeError: l.apply is not a function. (In 'l.apply(e,p)', 'l.apply' is undefined) - URL: http://octopi.local/static/webassets/packed_libs.js?86afb3b9 - Line: 133 - Column: 4373 - Error object: {}
2021-08-28 20:24:48,760 - octoprint.server.util.flask - INFO - Passively logging in user Sil from 2001:1c04:3f0b:6500:ac79:b06:2e9f:fc88
2021-08-28 20:24:48,762 - octoprint.access.users - INFO - Logged in user: Sil
2021-08-28 20:24:48,851 - octoprint.server.util.sockjs - INFO - Client connection closed: 2001:1c04:3f0b:6500:ac79:b06:2e9f:fc88
2021-08-28 20:24:49,174 - octoprint.server.util.sockjs - INFO - New connection from client: 2001:1c04:3f0b:6500:ac79:b06:2e9f:fc88
2021-08-28 20:24:49,199 - octoprint.server.util.flask - INFO - Passively logging in user Sil from 2001:1c04:3f0b:6500:ac79:b06:2e9f:fc88
2021-08-28 20:24:49,199 - octoprint.access.users - INFO - Logged in user: Sil
2021-08-28 20:24:51,507 - octoprint.plugins.printscheduler - DEBUG - Sorthing scheduled jobs from: [].
2021-08-28 20:24:51,507 - octoprint.plugins.printscheduler - DEBUG - Sorting scheduled jobs to: [].
2021-08-28 20:24:52,323 - octoprint.server.util.sockjs - INFO - User Sil logged in on the socket from client 2001:1c04:3f0b:6500:ac79:b06:2e9f:fc88
2021-08-28 20:24:59,746 - octoprint.plugins.printscheduler - DEBUG - Looping through scheduled jobs.
2021-08-28 20:25:59,748 - octoprint.plugins.printscheduler - DEBUG - Looping through scheduled jobs.
2021-08-28 20:26:59,751 - octoprint.plugins.printscheduler - DEBUG - Looping through scheduled jobs.
2021-08-28 20:27:59,753 - octoprint.plugins.printscheduler - DEBUG - Looping through scheduled jobs.
2021-08-28 20:28:59,756 - octoprint.plugins.printscheduler - DEBUG - Looping through scheduled jobs.
2021-08-28 20:29:59,758 - octoprint.plugins.printscheduler - DEBUG - Looping through scheduled jobs.
2021-08-28 20:30:59,759 - octoprint.plugins.printscheduler - DEBUG - Looping through scheduled jobs.
2021-08-28 20:31:59,762 - octoprint.plugins.printscheduler - DEBUG - Looping through scheduled jobs.

```

sildehoop, 2021-08-28:
browser.user_agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/14.1.2 Safari/605.1.15
connectivity.connection_check: 1.1.1.1:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1500
env.hardware.ram: 1903988736
env.os.bits: 32
env.os.id: linux
env.os.platform: linux
env.plugins.pi_support.model: Raspberry Pi 4 Model B Rev 1.2
env.plugins.pi_support.octopi_version: 0.18.0
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 20.3.3
env.python.version: 3.7.3
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.6.1
printer.firmware: Marlin bugfix-2.0.x (Nov 30 2020 12:25:54)
systeminfo.generator: systemapi

jneilliii, 2021-08-28:
are there any errors in your browser's developer tools (F12) within the console tab on page load?

sildehoop, 2021-08-28:
```
packed_plugins.js?bc317fcc:8942 Uncaught TypeError: element.datetimepicker is not a function
    at HTMLInputElement.<anonymous> (packed_plugins.js?bc317fcc:8942)
    at HTMLInputElement.dispatch (packed_libs.js?86afb3b9:13)
    at HTMLInputElement.v.handle (packed_libs.js?86afb3b9:13)
    at Object.trigger (packed_libs.js?86afb3b9:13)
    at HTMLInputElement.<anonymous> (packed_libs.js?86afb3b9:13)
    at Function.each (packed_libs.js?86afb3b9:13)
    at S.fn.init.each (packed_libs.js?86afb3b9:13)
    at S.fn.init.trigger (packed_libs.js?86afb3b9:13)
    at HTMLDivElement.<anonymous> (packed_plugins.js?bc317fcc:8930)
    at HTMLDivElement.dispatch (packed_libs.js?86afb3b9:13)
(anonymous) @ packed_plugins.js?bc317fcc:8942
dispatch @ packed_libs.js?86afb3b9:13
v.handle @ packed_libs.js?86afb3b9:13
trigger @ packed_libs.js?86afb3b9:13
(anonymous) @ packed_libs.js?86afb3b9:13
each @ packed_libs.js?86afb3b9:13
each @ packed_libs.js?86afb3b9:13
trigger @ packed_libs.js?86afb3b9:13
(anonymous) @ packed_plugins.js?bc317fcc:8930
dispatch @ packed_libs.js?86afb3b9:13
v.handle @ packed_libs.js?86afb3b9:13
trigger @ packed_libs.js?86afb3b9:13
(anonymous) @ packed_libs.js?86afb3b9:13
each @ packed_libs.js?86afb3b9:13
each @ packed_libs.js?86afb3b9:13
trigger @ packed_libs.js?86afb3b9:13
(anonymous) @ packed_plugins.js?bc317fcc:8930
dispatch @ packed_libs.js?86afb3b9:13
v.handle @ packed_libs.js?86afb3b9:13
packed_core.js?21da1722:853 ConfiguredLoggers has not changed. Not saving.
packed_libs.js?86afb3b9:133 Uncaught TypeError: l.apply is not a function
    at HTMLButtonElement.<anonymous> (packed_libs.js?86afb3b9:133)
    at HTMLButtonElement.dispatch (packed_libs.js?86afb3b9:13)
    at HTMLButtonElement.v.handle (packed_libs.js?86afb3b9:13)
```

sildehoop, 2021-08-28:
this is when I save

jneilliii, 2021-08-28:
please share your entire octoprint.log. I think there may be some possible conflict with another plugin in your case.

jneilliii, 2021-08-29:
Never mind. I can easily reproduce the issue. Not quite sure what's going on here, but will look into it.

jneilliii, 2021-08-29:
I think I have the issue fixed with the above commit and available as version 0.0.3rc2. You can change the plugin's release candidate in OctoPrint's Software Update settings to `Release Candidate` and upgrade when prompted to verify.

closed 2021-08-29 by sildehoop