jneilliii/OctoPrint-PrintScheduler #11: System Command/ GCODE

Zebsi235, 2021-08-30:
Hi, I'm relatively new to 3D Printing and really liked your idea of a scheduled print so I installed the Plugin. Now when I scheduled a print it wouldn't start automatically. My question now is if the reason for that is that I didn't fill in anything in the settings for: System Command Before Each Print, GCODE Before Each Print, ... . If that is the reason it doesn't start Printing then would you kindly tell me what to fill in those settings ( maybe the same lines that are in Cura ?).

Sorry if the question is too nooby haha ^^

jneilliii, 2021-08-30:
More than likely it has to do with the timezone/date time settings on your Pi not being correct. You need to SSH to your pi and set those using `sudo raspi-config`.

Zebsi235, 2021-08-30:
Ok I will look it up but the Upload Time and Date for Gcode Files is the right one on the Octoprint web UI. Does that differ with the raspberry's ?

jneilliii, 2021-08-30:
Yes, it can.

Zebsi235, 2021-08-30:
Alright looked it up and I think it really was the Problem. Thanks a LOT for your fast response and help. Appreciate every plugin of you :)

closed 2021-08-30 by Zebsi235