jneilliii/OctoPrint-PrintScheduler #17: Scheduled job was canceled but still printed.

BerikYoem, 2021-09-09:
Hi,

I'm using PrintScheduler 0.0.3 on Octoprint 1.6.1 running on my Windows10 laptop. 

I scheduled a print job (to start at 11:00 am) then removed it to start another print job manually. The manually started print job was finished and the print was still on the bed (9:15 am). I was away from the printer that time. Then the scheduled print job started at 11:00 am and it resulted the hotend to crush onto the print that was still on the bed. I was able to see what happened because I set up a IP security camera to watch my printer and by checking the Octoprint log. 

The PrintScheduler started the print job at 11:00 am even though I have already removed it from the schedule list.

jneilliii, 2021-09-09:
After removing from the list did you press the save button?

BerikYoem, 2021-09-11:
I can't remember if I pressed the save button after removing the task from the scheduler, however, is the Save button necessary after removing a task from the scheduler? There's no re-do button to bring a task back after removing it anyway so I think it's gonna be more instinct once a task is removed it's removed without Saving.

jneilliii, 2021-09-11:
Yeah, until save is pressed the actual queue is only modified locally in the web ui. Auto save occurs if you clear all, but not on single removal.

BerikYoem, 2021-09-12:
I think that auto save should occur on a single removal as well since there's no going back option when a print job is removed from the schedule list. At the moment it's like just adding another layer of meaningless procedure to do a task.

jneilliii, 2021-09-12:
except, you can press refresh on the page to bring back the saved list. I could see where this could be confusing.

jneilliii, 2021-10-02:
changed with above commit. and released with version 0.0.5rc2. If you'd like to test it out change the plugin's release channel in OctoPrint's Software Update settings to `Release Candidate` and upgrade when prompted.

Deses, 2022-01-12:
I tested version 0.0.5rc3 and this is working fine for me (Raspi 4 4Gb with OctoPrint 1.7.2 / Python 3.7.3 / OctoPi 0.17.0).

It's definitely more intuitive now. Thanks!

PR #36 (2022-10-15): 1.0.0
* add additional check on print start event to avoid error, resolves #20
* save settings when removing single file, resolves #17
* add option for repeating scheduled jobs daily, resolves #22
* added navbar button to indicate if there are scheduled print jobs or not, clicking it will take you to the print scheduler tab
* adding jobs to print scheduler will automatically switch to plugin's tab
closed 2022-10-15 by jneilliii