jneilliii/OctoPrint-PrintScheduler #19: Could this work with a TP Link power switch?

BixlerVDL, 2021-09-21:
I think it would be an even better feature if it could turn the printer on just before each print, because now I have to leave the printer idle.

Thanks for your time, it already is such a good feature thanks :)

jneilliii, 2021-09-21:
It can with my TPLink plugin. Check out the wiki [here](https://github.com/jneilliii/OctoPrint-PrintScheduler/wiki/Helpful-System-Commands).

closed 2021-09-21 by jneilliii

debren27, 2022-04-26:
@jneilliii I recommend you link to the Helpful System Commands wiki page from the wiki Home Page. I went looking for this info on the wiki but the wiki looked blank; I didn't see the tree on the right. I found the info here in this closed Issue but only because I was trying to find the info myself and not bother you with an FAQ, but ironically now here I am bothering you.

jneilliii, 2022-04-26:
I can see how it may not be very visible with it auto-collapsed on the right. I've added a line to the main wiki page, thanks for the suggestion.