jneilliii/OctoPrint-PrintScheduler #20: AttributeError: 'NoneType' object has no attribute 'cancel'

ThomasCr, 2021-09-23:
I got this on my logfiles.
I use 

- OctoPrint 1.6.1
- Python 3.8.10
- Print Scheduler[printscheduler] v0.0.4

```
2021-09-23 07:49:24,602 - octoprint.util.comm - INFO - Changing monitoring state from "Starting" to "Printing"
2021-09-23 07:49:24,614 - octoprint.plugin - ERROR - Error while calling plugin printscheduler
Traceback (most recent call last):
  File "/opt/octoprint/lib/python3.8/site-packages/octoprint/plugin/__init__.py", line 271, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/opt/octoprint/lib/python3.8/site-packages/octoprint/util/__init__.py", line 1941, in wrapper
    return f(*args, **kwargs)
  File "/opt/octoprint/lib/python3.8/site-packages/octoprint_printscheduler/__init__.py", line 73, in on_event
    self.repeated_timer.cancel()
AttributeError: 'NoneType' object has no attribute 'cancel'
```

jneilliii, 2021-09-23:
Please upload full octoprint.log so I can see the error in context and know what might be causing it.

ThomasCr, 2021-09-23:
I am not sure if it will really help...

```
2021-09-23 07:48:47,401 - octoprint.plugins.tracking - INFO - Sent tracking event printer_connected, payload: {'firmware_name': 'Marlin 1.1.9 (Github)', 'pri
nter_port': 'AUTO', 'printer_baudrate': 115200}
2021-09-23 07:49:24,348 - octoprint.filemanager.analysis - INFO - Starting analysis of local:LLK5PRO_bltouch-mount.gcode
2021-09-23 07:49:24,349 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /opt/octoprint/bin/python -m octoprint analysis gcode --speed-x=
6000 --speed-y=6000 --max-t=10 --throttle=0.0 --throttle-lines=100 --bed-z=0.0 /opt/octoprint/.octoprint/uploads/LLK5PRO_bltouch-mount.gcode
2021-09-23 07:49:24,350 - octoprint.plugin - ERROR - Error while calling plugin calibrationcompanion
Traceback (most recent call last):
  File "/opt/octoprint/lib/python3.8/site-packages/octoprint/plugin/__init__.py", line 271, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/opt/octoprint/lib/python3.8/site-packages/octoprint/util/__init__.py", line 1941, in wrapper
    return f(*args, **kwargs)
  File "/opt/octoprint/lib/python3.8/site-packages/octoprint_calibrationcompanion/__init__.py", line 226, in on_event
    if payload['name'] == self.filename + ".gcode":
AttributeError: 'calibrationcompanion' object has no attribute 'filename'
2021-09-23 07:49:24,436 - octoprint.printer.standard.job - INFO - Print job selected - origin: local, path: LLK5PRO_bltouch-mount.gcode, owner: thomas, user:
 thomas
2021-09-23 07:49:24,553 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Starting"
2021-09-23 07:49:24,574 - octoprint.printer.standard.job - INFO - Print job started - origin: local, path: LLK5PRO_bltouch-mount.gcode, owner: thomas, user:
thomas
2021-09-23 07:49:24,581 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2021-09-23 07:49:24,602 - octoprint.util.comm - INFO - Changing monitoring state from "Starting" to "Printing"
2021-09-23 07:49:24,614 - octoprint.plugin - ERROR - Error while calling plugin printscheduler
Traceback (most recent call last):
  File "/opt/octoprint/lib/python3.8/site-packages/octoprint/plugin/__init__.py", line 271, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/opt/octoprint/lib/python3.8/site-packages/octoprint/util/__init__.py", line 1941, in wrapper
    return f(*args, **kwargs)
  File "/opt/octoprint/lib/python3.8/site-packages/octoprint_printscheduler/__init__.py", line 73, in on_event
    self.repeated_timer.cancel()
AttributeError: 'NoneType' object has no attribute 'cancel'
2021-09-23 07:49:24,727 - octoprint.plugins.tracking - INFO - Sent tracking event print_started, payload: {'origin': 'local', 'file': '18645861c6ba5710f7fb90de515b6696adfce9ee'}
2021-09-23 07:49:26,651 - octoprint.util.comm - INFO - Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
2021-09-23 07:52:01,627 - octoprint.util.comm - INFO - Telling the printer to set the busy interval to our "communicationBusy" timeout - 1s = 2s
2021-09-23 07:57:20,750 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-09-23 07:57:29,105 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 4509, 'printer_state': 'PRINTING'}
2021-09-23 08:12:20,751 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-09-23 08:12:29,148 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 5409, 'printer_state': 'PRINTING'}
2021-09-23 08:27:20,752 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-09-23 08:27:29,119 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 6309, 'printer_state': 'PRINTING'}
2021-09-23 08:31:46,092 - octoprint.util.comm - INFO - Finished in 2541.554 s.
2021-09-23 08:31:46,092 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Finishing"
```

jneilliii, 2021-09-23:
Yeah, not much help. I'll look into it. Probably just need to add a little if check in there. I assume that the file listed as started printing was not added to the scheduled jobs or was it?

jneilliii, 2021-09-23:
Basically I need the steps required to reproduce the error.

jneilliii, 2021-09-24:
I've added an additional check in the above commit and released in version 0.0.5rc1. If you'd like to try it out you can change the plugin's release channel in OctoPrint's Software Update settings to `Release Candidate` and upgrade when prompted.

Deses, 2022-01-12:
How did you reproduce the bug? While I'm using version 0.0.5rc3 I might give it a try to give you some feedback given that OP has forgot about this...

jneilliii, 2022-01-12:
I purely made a guess based on the error message. I have no clue how to reproduce the issue.

ivebeenlinuxed, 2022-03-28:
I think this error comes from a pause? If you pause, then start again, then the self.repeated_timer will be None and therefore cannot cancel.

jneilliii, 2022-03-28:
Thanks @ivebeenlinuxed are you running the 0.0.5rc3 version? I added checks in 0.0.5rc1 that should have resolved this issue.

PR #36 (2022-10-15): 1.0.0
* add additional check on print start event to avoid error, resolves #20
* save settings when removing single file, resolves #17
* add option for repeating scheduled jobs daily, resolves #22
* added navbar button to indicate if there are scheduled print jobs or not, clicking it will take you to the print scheduler tab
* adding jobs to print scheduler will automatically switch to plugin's tab
closed 2022-10-15 by jneilliii