jneilliii/OctoPrint-PrintScheduler #21: octoprint.plugins.printscheduler - DEBUG - Bypassing scheduled job as printer is not available yet.

LexHouben, 2021-09-30:
Hi 
I can turn the printer on and off via Tasmota.
Via curl 'http://192.168.XX.XX/cm?cmnd=Power On' I can turn it on in the CLI.
However, this does not work in combination with Print Scheduler.
What are the correct settings to turn on the printer, via tasmota, and then let the print scheduler start printing?
When the printer is online, 'Print Scheduler' works as it should.

jneilliii, 2021-09-30:
The message indicates printer is not automatically connecting. My tasmota plugin does this for you when you use the auto connect feature. Otherwise, you'll need something like the portlister plugin configured to automatically connect when your primer powers on.

closed 2021-09-30 by jneilliii

LexHouben, 2021-10-01:
When I switch the tasmota it connects automatic.
[image]
_2021-09-30 12:52:03,026 - octoprint.plugins.printscheduler - DEBUG - Looping through scheduled jobs.
2021-09-30 12:52:03,028 - octoprint.plugins.printscheduler - DEBUG - Job found: {'name': 'Body2_0.3mm_PLA_MK3SMMU2S_6h10m.gcode', 'path': 'Body2_0.3mm_PLA_MK3SMMU2S_6h10m.gcode', 'start_at': '2021-09-30 12:50'}
2021-09-30 12:52:03,028 - octoprint.plugins.printscheduler - DEBUG - Running system command before print.
2021-09-30 12:52:03,304 - octoprint.plugins.printscheduler - DEBUG - Bypassing scheduled job as printer is not available yet.
2021-09-30 12:53:03,305 - octoprint.plugins.printscheduler - DEBUG - Looping through scheduled jobs.
2021-09-30 12:53:03,306 - octoprint.plugins.printscheduler - DEBUG - Job found: {'name': 'Body2_0.3mm_PLA_MK3SMMU2S_6h10m.gcode', 'path': 'Body2_0.3mm_PLA_MK3SMMU2S_6h10m.gcode', 'start_at': '2021-09-30 12:50'}
2021-09-30 12:53:03,307 - octoprint.plugins.printscheduler - DEBUG - Running system command before print.
2021-09-30 12:53:03,489 - octoprint.plugins.printscheduler - DEBUG - Bypassing scheduled job as printer is not available yet._

But when I look in the log file of the print scheduler I suspect that the print scheduler is checking if the printer is connected. If this is not the case, it waits for it to come online and only then starts printing.
This is also the case because if the print is not executed and I turn on the printer, via the tasmota switch, it immediately starts printing.
In the log file it also gives the message that the PCB is not connected and waits for the next check.

jneilliii, 2021-10-01:
That is exactly what it does. If printer is not in operational state it picks it up in the next loop. Any errors in your logs?