jneilliii/OctoPrint-PrintScheduler #22: Is it possible to assign time without date?

fly115, 2021-10-01:
Great work. Thankyou. 
Would be great if you could assign a time without a date. Or if there was a checkbox for 'repeat daily'
Looking a way to have a Sisyphus style sand table tell the time by drawing a clock face and then moving the ball based on time. I can divide up the time movements into separate GCode files but would need to schedule each file to 'print' dependent on the time (but not date).

jneilliii, 2021-10-01:
You should check out my unlisted plugin OctoPrint-Playlist, it was designed specifically for sand printers....

https://github.com/jneilliii/OctoPrint-Playlist

jneilliii, 2021-10-01:
technically you could probably do it with a single repeating file using G4 dwell commands

fly115, 2021-10-01:
Yes I have tried your OctoPrint-Playlist plugin too. Thank you. Only problem with using it is that you can only assign a start time to the whole playlist not to individual files. Multiple playlists would work (if that were possible). 
So yes I'm currently working in the direction of using dwell commands and then just kick off the print at right time. I suspect it will slowly get out of time through the day. 
I guess my hope was that i could kick it off at any time and it would jump straight to the correct time (using schedule to determine which file to 'print').

jneilliii, 2021-10-01:
Well, even with this plugin the time could be off by a minute or two.

jneilliii, 2021-10-02:
added `repeat daily` feature in above commit. and released with version 0.0.5rc2. If you'd like to test it out change the plugin's release channel in OctoPrint's Software Update settings to `Release Candidate` and upgrade when prompted. It should in theory be close to what you want, but remember the time may not be perfect as the way the plugin works is fairly remedial date/time comparisons and are not tied to an actual clock, or cron jobs or anything like that.

jneilliii, 2021-10-02:
BTW, would love to get a timelapse of your sand clock if you do get it working with these changes.

fly115, 2021-10-02:
Wow thanks so much. I am really grateful. I'll play around with it and let you know how things go. I don't need anything super accurate because the resolution of the ball movement is not high anyway. Im hoping I can use this to sync up time occasionally and go to the correct time on startup. If it all works out I'd be happy to share a time lapse and gcodes

PR #36 (2022-10-15): 1.0.0
* add additional check on print start event to avoid error, resolves #20
* save settings when removing single file, resolves #17
* add option for repeating scheduled jobs daily, resolves #22
* added navbar button to indicate if there are scheduled print jobs or not, clicking it will take you to the print scheduler tab
* adding jobs to print scheduler will automatically switch to plugin's tab
closed 2022-10-15 by jneilliii