jneilliii/OctoPrint-PrintScheduler #26: Suggestion: Checkbox to override operational state.

phantom101, 2022-01-27:
First a big thank you for making this nice plugin !

But sadly its not useable for me and maybe some other people because there is no way to override the operational state check.

Why:
I use the PSU Plugin to switch of the power of my printer when its not in use (relay) so the Printer is not in an operational state and the sheduled print will not be executed. There is allready a option to send a G Code or Systemcommand like "psucontrol:on" before starting the print that would my printer turn on and bring it to a operational state but this command is not  executed because the printer is not in operational state ...

So it would be nice to have a checkbox to override the check ...
Or the better way a checkbox that the "before printing" G Code and Systemcommands are executed without a check then wait lets say 10 seconds and after that do the operational state check and start printing.

Thank you very much !
Patrick

jneilliii, 2022-01-27:
You mean powering on with psu control prior to starting? Maybe this will help? 

https://github.com/jneilliii/OctoPrint-PrintScheduler/wiki/Helpful-System-Commands

phantom101, 2022-01-27:
Thank You !!!
And sorry that i have not seen the wiki article before i post :-(
Now its working :-)

closed 2022-01-27 by jneilliii