jneilliii/OctoPrint-PrintScheduler #29: Having issues with powering on thru method mentioned in fixed issue

Voxel3DPrinting, 2022-05-12:
Hi,

I ain't much of a programmer myself, therefore my (maybe stupid) question, but I cannot get the auto power on to work before each scheduled print.
I do not understand other articles on what exactly I need to do but based on the article "https://github.com/jneilliii/OctoPrint-PrintScheduler/wiki/Helpful-System-Commands" I tried adding "/home/pi/oprint/bin/octoprint plugins psucontrol:on" to the "System Command Before Each Print" section. Unfortunately, it did not work. I do know for sure I have the PSU Control plugin installed which works fine.

Is anyone feeling generous to help??
Thanks in advance!

jneilliii, 2022-05-12:
I might have to get the assistance of @kantlivelong on this one, as I do not know much in regard to how the PSU Control plugin's CLI commands work.

kantlivelong, 2022-05-13:
The CLI just makes API calls so it's not much different than using `curl`. Was it tested manually from SSH?

Voxel3DPrinting, 2022-05-14:
> The CLI just makes API calls so it's not much different than using `curl`. Was it tested manually from SSH?

I do not have the knowledge how to test it with SSH or what SSH exactly is in the first place.

I just would like to know which steps to take to get the printer turned on before a print job, with the printschedule and PSUcontrol plugin.

If you could tell me what to do I want to help

jneilliii, 2022-05-15:
how to SSH: https://libguides.nyit.edu/c.php?g=469894&p=3365470

once there run the same command you entered in the plugin's settings and see what's returned.

Voxel3DPrinting, 2022-05-20:
I asked some someone with knowledge and will try that this evening

Voxel3DPrinting, 2022-05-20:
Hi, I tried what you said and sent the command thru putty SSH connection, it returns "no such file or directory"
image is attached.
[image]

kantlivelong, 2022-05-20:
Replace `pi` with the username you chose to use.

Voxel3DPrinting, 2022-05-20:
this worked, thanks!
I added this line to the start command. Maybe an Idea to add this instruction to the wiki page? or mention this issue??

The command now works, but the print does not start at the given time. I think the time is set incorrectly. Is there any way to check the time raspberry thinks it is?

jneilliii, 2022-05-20:
Check your logs, they are time stamped. You probably have to ssh to the pi and set timezone.

jneilliii, 2022-05-20:
added note to wiki page

Voxel3DPrinting, 2022-05-20:
> Check your logs, they are time stamped. You probably have to ssh to the pi and set timezone.

They seem to report the right time, I live in the Netherlands
[image]

Any way I can check it in SSH?

Voxel3DPrinting, 2022-05-20:
> added note to wiki page

Cool!

kantlivelong, 2022-05-20:
> Any way I can check it in SSH?

`timedatectl`

Voxel3DPrinting, 2022-05-22:
I did it! went to sudo raspi-config, then to localisation options-> timezone and to my local timezone

Should work now! thanks so much

Time reports good

closed 2022-05-23 by jneilliii