jneilliii/OctoPrint-PrintScheduler #33: Scheduled print starting although is was deleted

remcoder, 2022-07-31:
First of all, I want to thank you for this plugin (and your other plugins)!! It's amazing how much and I am very grateful for all time and effort you've put into this ;-) 

Today I ran into a very weird situation, basically boiling down to the fact that a scheduled print started although I had already deleted the scheduled job.

Here are the steps:
- I scheduled my Printer to switch on via a Philips Hue smart plug at 8:15. I verified this happened on time. 
- I ssh'ed into my pi to check the time. I verified via the `date` command that the reported time equals actual time.
- I scheduled a print to start at 8:30 July 31st. This didn't happen.
- At about 8:35 I started the print manually via Octoprints interface.
- At some point while the print was running I removed the still scheduled print via the UI
- After the print was done, which took about 2 hours, the scheduled print was starting automatically. 
- At this point I reset the printer manually to get out of the print and not hit the print on the bed.

I think 2 things happened:
- The scheduled print did not start at 8:30 because of a timezone or time formatting issue. In fact, Octoprint's logs are off by an hour. I probably should file a separate issue with Octoprint for that.
- The scheduled print was not removed properly from the queue although I removed it via the UI. If true then this is a bug in the plugin. 

[octoprint.log](https://github.com/jneilliii/OctoPrint-PrintScheduler/files/9228575/octoprint.log)

I'll try to investigate some more and update this issue with any findings.

jneilliii, 2022-07-31:
The timezones is more than likely the issue for the time problem. SSH to the pi and use `sudo raspi-config` localization options to adjust appropriately. For the second one I'm not actually sure I debugged/tested removing a scheduled job but I'm fairly confident I did. Curious if the timezone setting being off could potentially break the removal logic in the plugin. Let's fixed that first and see if it fixes the other.

remcoder, 2022-07-31:
Thanks for the suggestion. I updated my timezone and now the Octoprint logs are in sync with my local time. And like you suspected the issue didn't occur anymore even though I tried to reproduce the exact scenario.

I guess I didn't reboot the pi the first time I changed the timezone which left Octoprint with the old timezone setting.

closed 2022-07-31 by remcoder