jneilliii/OctoPrint-PrintScheduler #34: No Schedule Button?

3DCoded, 2022-09-06:
I have installed PrintScheduler and I don't see a clock icon on my files. I am using UI Customizer Plugin too. Please help!
[image]

jneilliii, 2022-09-06:
Probably a conflict with another plugin, what is the plus sign button?

3DCoded, 2022-09-06:
It's Continuous Print.

jneilliii, 2022-09-06:
If you disable that plugin do you then see the schedule button?

3DCoded, 2022-09-06:
Currently I am in a print job, so I will let you know what happens after it's done. Is there a way to select the goode file directly from the PrintScheduler tab?

jneilliii, 2022-09-06:
No, unfortunately not.

3DCoded, 2022-09-06:
Hmm... Now the button shows when I disable the plugin, but when I click it doesn't do anything. For now, I'm going to re-enable the continuous print plugin until this is fixed.

jneilliii, 2022-09-06:
That doesn't make sense, did you restart OctoPrint and refresh the page after disabling?

jneilliii, 2022-09-06:
Just realized, the current stable version doesn't automatically switch you to the Print Scheduler tab when adding a print to it. You have to manually switch to it. The latest release candidate version automatically switches.

3DCoded, 2022-09-07:
Is there a way to have Continuous Print and PrintScheduler running simultaneously?

ManeLippert, 2022-12-30:
I can confirm that disabling Continuous Print plugin and restart Octoprint brings back the Print Scheduler plugin. The plugin works than just as intended with auto switch. Reenable Continuous Print let's Scheduler button disappear.

jneilliii, 2022-12-31:
Should be fixed in the latest 1.0.1 release.

closed 2022-12-31 by jneilliii