jneilliii/OctoPrint-PrintScheduler #35: Problem with connection

DarthWeber, 2022-09-07:
Hi and thanks for this great plugin. But i have the problem, that the print doen't start because it is disconnected. I need to press the connect button in octoprint and the the scheduled job starts immediatly.

Did i missed something ?

reagrds,
DW

jneilliii, 2022-09-07:
OctoPrint's connection to the printer must be active for scheduled jobs to run. If the printer isn't powered on or in a disconnected state then the plugin won't start a print. Also, you need to make sure the localization options are updated for your timezone on the pi for the correct time to be used for starting the print.

DarthWeber, 2022-09-12:
sorry, sorry, sorry - it was my mistake. I used a different port for octoprint as 80 was already in use. Thats the reason i didn't got any error - i simply forgot it.

closed 2022-09-12 by DarthWeber