jneilliii/OctoPrint-PrintScheduler #38: Prints not starting

Ben-Bartlett, 2022-11-02:
I'm currently using a raspberry pi 4b with octopront version 1.8.6, octopi version 0.18.0. It is connected to a prusa i3 mk3s+.

I've installed the plugin with the manager and via the url. 

I have checked the time settings on the pi with sudo raspi-config.
[image]
No print starts when the scheduled time arrives. Any ideas?

Log file attached
[octoprint.log](https://github.com/jneilliii/OctoPrint-PrintScheduler/files/9919286/octoprint.log)

jneilliii, 2022-11-02:
so both timezone and date/time I configured correctly? double-check that the items are getting added to config.yaml when you add them to the scheduler queue. nothing in log indicates there's an error.

Ben-Bartlett, 2022-11-02:
I've double checked the timezone again and I have set it to my timezone. 

below the print seems to be added to the config.yaml file but the print still doesn't start up
[image]

jneilliii, 2022-11-02:
I forgot to request that you enable debug logging for the plugin in the logging section of OctoPrint. That will then write lines out to octoprint.log correctly to debug what's going on. Select octoprint.plugins.printscheduler from the list and click save. 

[image]

Ben-Bartlett, 2022-11-02:
[octoprint (2).log](https://github.com/jneilliii/OctoPrint-PrintScheduler/files/9923895/octoprint.2.log)

jneilliii, 2022-11-03:
ah...are your file stored on SD card and not in local OctoPrint storage? I never tested that and is probably the reason I think this might be happening.

Ben-Bartlett, 2022-11-03:
yes the files were on the sd. Switched to the octoprint storage and the prints start up fine. Thanks very much