jneilliii/OctoPrint-PrintScheduler #41: Installation issue "SetuptoolsDeprecationWarning: setup.py install is deprecated"

elpeter25, 2023-01-03:
Hi,

I'm trying to install the pluging and I'm having following error:

```
error: subprocess-exited-with-error
× Running setup.py install for backports-datetime-fromisoformat did not run successfully.
│ exit code: 1
╰─> [25 lines of output]
/home/octoprint/python_venvs/octoprint1/lib/python3.7/site-packages/setuptools/version.py:1: UserWarning: Module backports was already imported from None, but /tmp/pip-install-hp3a2ol6/backports-datetime-fromisoformat_a9a7a595b31448e4977bb85b05e7e20d is being added to sys.path
import pkg_resources
running install
/home/octoprint/python_venvs/octoprint1/lib/python3.7/site-packages/setuptools/command/install.py:37: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.setuptools.SetuptoolsDeprecationWarning,
running build
running build_py
creating build
creating build/lib.linux-x86_64-cpython-37
creating build/lib.linux-x86_64-cpython-37/backports
copying backports/__init__.py -> build/lib.linux-x86_64-cpython-37/backports
creating build/lib.linux-x86_64-cpython-37/backports/datetime_fromisoformat
copying backports/datetime_fromisoformat/__init__.py -> build/lib.linux-x86_64-cpython-37/backports/datetime_fromisoformat
running build_ext
building 'backports._datetime_fromisoformat' extension
creating build/temp.linux-x86_64-cpython-37
creating build/temp.linux-x86_64-cpython-37/backports
creating build/temp.linux-x86_64-cpython-37/backports/datetime_fromisoformat
x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I/home/octoprint/python_venvs/octoprint1/include -I/usr/include/python3.7m -c backports/datetime_fromisoformat/_datetimemodule.c -o build/temp.linux-x86_64-cpython-37/backports/datetime_fromisoformat/_datetimemodule.o
In file included from backports/datetime_fromisoformat/_datetimemodule.
c:66:0:
backports/datetime_fromisoformat/_datetimemodule.h:1:10: fatal error: Python.h: No such file or directory
#include <Python.h>
^~~~~~~~~~
compilation terminated.
error: command '/usr/bin/x86_64-linux-gnu-gcc' failed with exit code 1
[end of output]
note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure
× Encountered error while trying to install package.
╰─> backports-datetime-fromisoformat
note: This is an issue with the package mentioned above, not pip.hint: See above for output from the failure.
```

I'm using OctoPrint 1.8.6 and Python 3.7.15

jneilliii, 2023-01-03:
The message seems to indicate that you are using a manual install of OctoPrint and what instructions/tool did you use to do so? Could you let me know what the base OS is for further troubleshooting? More than likely it's an upstream dependency missing.

elpeter25, 2023-01-13:
That's correct. I'm using Ubuntu 18 as OS.

Linux version 4.15.0-196-generic (buildd@lcy02-amd64-018) (gcc version 7.5.0 (Ubuntu 7.5.0-3ubuntu1~18.04)) #207-Ubuntu SMP Thu Oct 27 21:24:58 UTC 2022

DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=18.04
DISTRIB_CODENAME=bionic
DISTRIB_DESCRIPTION="Ubuntu 18.04.6 LTS"

jneilliii, 2023-01-13:
The pip install error is a new thing that has recently come up. Mu understanding is that the next release of OctoPrint will resolve this issue, but I think it's just a matter of pip installing wheel from the venv folder. Not so certain about the older version of Ubuntu though. @paukstelis have you been seeing this with the install/deploy scripts?

paukstelis, 2023-01-13:
Yes, this is a really a python version thing. I think 18.04 maxes out at 3.6 or 3.7 and OctoPrint now requires 3.8 (I think it is). install/deploy specifies Ubuntu20+ for this reason.

jneilliii, 2023-01-13:
Yeah, OctoPrint is set for >=3.7 for python, so if 3.6 is what's there then that's a larger problem for sure. I see from the venv lines in the log above though that 3.7 is what Peter has installed.

elpeter25, 2023-01-17:
> Yeah, OctoPrint is set for >=3.7 for python, so if 3.6 is what's there then that's a larger problem for sure. I see from the venv lines in the log above though that 3.7 is what Peter has installed.

Yes it is indeed 3.7 python

jneilliii, 2023-01-17:
Quick googling on the error indicates that possible running `sudo apt-get install python3-dev` might fix the issue.

jneilliii, 2023-01-17:
you may have to tell it the version though with `sudo apt-get install python3.7-dev`.

jneilliii, 2023-01-17:
and if neither of those help `sudo apt-get install build-essential` may get you back in working order.

elpeter25, 2023-01-27:
> Quick googling on the error indicates that possible running `sudo apt-get install python3-dev` might fix the issue.

```
Reading package lists... Done
Building dependency tree
Reading state information... Done
python3-dev is already the newest version (3.6.7-1~18.04).
0 upgraded, 0 newly installed, 0 to remove and 3 not upgraded.
```

> you may have to tell it the version though with `sudo apt-get install python3.7-dev`.

That worked like a charm!!! 

Thanks a bunch!!

elpeter25, 2023-01-27:
Thank yo uso much @jneilliii for your time. It worked after installing python-dev with the correct version!!

closed 2023-01-27 by elpeter25