jneilliii/OctoPrint-PrintScheduler #43: Time picker backwards interaction with the mouse wheel.

Deses, 2023-03-19:
The date and time picker looks like is inverted when using the mouse wheel. Let me explain:

00:00 is at the top of the picker and 23:55 is at the bottom. If you click the little arrows on top/bottom, it goes up/down as expected, if you drag the scrollbar it also behaves as I would expect. **Then the non intuitive part (for me at least)**: if I scroll down, time goes down towards 00:00 but the picker and the scroll bar go up, it feels like it's backwards.

If I go to the documentation and examples of the datetime picker you used (https://xdsoft.net/jqplugins/datetimepicker/) their example feels fine.

I understand the logic behind `mouse wheel down` = decreasing time, but I think it's not intuitive and it should behave like the examples.

jneilliii, 2023-03-19:
I actually don't know why it does it reversed like that in the plugin, I'm not overriding any settings or defining the behavior directly. I suppose it might be the [knockout binding](https://github.com/jneilliii/OctoPrint-PrintScheduler/blob/master/octoprint_printscheduler/static/js/ko.datetimepicker.js) I'm using, but don't see anything specific to why it's rolling the way it does.

jneilliii, 2023-03-19:
Seems like this was fixed in a later version of the library that I used. I've updated the library and mouse scroll direction seems right now in version 1.0.2.

closed 2023-03-19 by jneilliii

Deses, 2023-03-19:
Excellent! Thank you for your work.