jneilliii/OctoPrint-PrintScheduler #44: Question about print scheduler

mich0111, 2023-05-29:
Hello,
May I know the list and forms of commands which could be launch before and after print in the parameters of plugin ?
Thx in advance
Michel

jneilliii, 2023-05-29:
This is a very broad question. The gcode section can run any command that your printer supports, a good reference would be [here](https://reprap.org/wiki/G-code). The system commands section in settings will depend on the OS that OctoPrint is installed on, but in general can run any command that your OS supports.

mich0111, 2023-05-29:
Hi,

Thank you for your quick reply.
For the OS and gcode commands, I can imagine it, but I thought there could,
for instance, be commands to switch off or on the printer, the light, i.e.
other commands than those mentioned.

Best regards

Michel CHEMLA

jneilliii, 2023-05-29:
oh yeah, some of the known plugins that control power have examples. don't really have any more than what's documented [here](https://github.com/jneilliii/OctoPrint-PrintScheduler/wiki/Helpful-System-Commands) in the wiki.

closed 2023-06-21 by github-actions[bot]