jneilliii/OctoPrint-PrintScheduler #45: Could not import OctoPrint's setuptools

narcolepsyboy, 2023-09-27:
Hi there, 

Hope you can help. I am a newbie to octoprint, I have just installed the latest version of it and would like to isntall your scheduler but am getting the below. Apologies if it's obvious! 

 OctoPrint 1.9.2 Python 3.9.2 OctoPi* 1.0.0 (build 2023.07.20.144556) on a Pi Zero W 2

 sudo pip install "https://github.com/jneilliii/OctoPrint-PrintScheduler/archive/master.zip"
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/jneilliii/OctoPrint-PrintScheduler/archive/master.zip
  Downloading https://github.com/jneilliii/OctoPrint-PrintScheduler/archive/master.zip
     / 225 kB 403 kB/s
    ERROR: Command errored out with exit status 255:
     command: /usr/bin/python3 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-mq77wcmm/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-mq77wcmm/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-ajkc5g4p
         cwd: /tmp/pip-req-build-mq77wcmm/
    Complete output (1 lines):
    Could not import OctoPrint's setuptools, are you sure you are running that under the same python installation that OctoPrint is installed under?
    ----------------------------------------
WARNING: Discarding https://github.com/jneilliii/OctoPrint-PrintScheduler/archive/master.zip. Command errored out with exit status 255: python setup.py egg_info Check the logs for full command output.
ERROR: Command errored out with exit status 255: python setup.py egg_info Check the logs for full command output.

An issue with my python version? I have upgraded and updated the OS.

Many thanks :)

jneilliii, 2023-09-28:
> sudo pip install "https://github.com/jneilliii/OctoPrint-PrintScheduler/archive/master.zip"

This looks wrong. Why are you trying to manually install from command line as sudo?

narcolepsyboy, 2023-09-28:
I was getting the same error installing normally so thought to try sudo,
that was the only reason. I also tried via the Octo print gui plugins
section but it fails there too.

jneilliii, 2023-09-28:
I need the original errors from the attempts installing from GUI. Please upload the octoprint.log and plugin manager console log.

narcolepsyboy, 2023-09-29:
Hi there

I just tried again yesterday and strangely this time it worked! All good,
sorry to bother you. I used it last night worked beautifully!

Thanks again for your help and great plugin!

closed 2023-09-29 by jneilliii