jneilliii/OctoPrint-PrintScheduler #49: Pre print command not executed

FajnB, 2024-06-23:
**I am running windows version of OctoPrint**

When I tried to run commands from "System Command Before Each Print" the commands are not run.
I wanted to have the printer off when it is not printing, and as soon as print schedule hits, it turns on the smart socket.
At first I thought it was the curl problem with Windows so I tested it with Invoke-WebRequest:
```Invoke-WebRequest -Uri "http://<OCTOPRINT_IP>/api/plugin/tasmota" -Method Post -Headers @{"Content-Type"="application/json";"X-Api-Key"="<API_KEY>"} -Body (@{command="turnOn";ip="<IP>";idx="<IDX>"} | ConvertTo-Json)```
This didn't work either. So I tried the last thing and use the solution from #18 with Invoke-WebRequest:
```Invoke-WebRequest -s http://<TASMOTA_IP>/cm?cmnd=Power%20On```

Did not work either.
The Invoke-WebRequest commands work as they should in PowerShell.
And when looking through the logs it seems that the plugin sees that there is a command and tries to run it but nothing really happens.

Log:
```2024-06-23 11:22:40,485 - octoprint.plugins.printscheduler - DEBUG - Looping through scheduled jobs.
2024-06-23 11:22:40,485 - octoprint.plugins.printscheduler - DEBUG - Job found: {'name': 'TT_firstlayer_2_w_purge_line.gcode', 'path': 'TT_firstlayer_2_w_purge_line.gcode', 'start_at': '2024-06-23 11:22'}
2024-06-23 11:22:40,486 - octoprint.plugins.printscheduler - DEBUG - Running system command before print.
2024-06-23 11:22:40,520 - octoprint.plugins.printscheduler - DEBUG - Bypassing scheduled job as printer is not available yet.```

jneilliii, 2024-06-23:
Commands must include full path to executable. Just putting in powershell command won't work, you need to run powershell exe for that. Same for curl command, there needs to be full path to the exe.

FajnB, 2024-06-23:
Oh. That makes ton of sense.
I propose than a slight tweak in your documentation so that people not as good with IT can follow and proceed to make functional system commands calls.

What I mean is that you should add a _**Note**_ that system command has to have full path to the terminal in which should the command be run in.
For example on linux _**bash**_ and on windows _**powershell -command "YOUR_COMMAND"**_ or _**cmd.exe /c YOUR_COMMAND**_

Also thanks as now the command works as intended:
`powershell -command "Invoke-WebRequest -Uri 'http://<OCTOPRINT_IP>/api/plugin/tasmota' -Method Post -Headers @{'Content-Type'='application/json';'X-Api-Key'='<API_KEY>'} -Body (@{command='turnOn';ip='<IP>';idx='<IDX>'} | ConvertTo-Json)"`